adafruit/Adafruit_CircuitPython_MCP3xxx #2: Add Support for MCP3002

brentru, 2018-07-20:
Support for the MCP3002: 2.7V Dual Channel 10-Bit A/D Converter with SPI Serial Interface

Datasheet: http://ww1.microchip.com/downloads/en/DeviceDoc/21294E.pdf

2bndy5, 2019-10-09:
I have limited access to a couple MCP3002, so I plan on addressing this issue when I snag them from my robotics club's supply. I have [started a branch](https://github.com/2bndy5/Adafruit_CircuitPython_MCP3xxx/tree/mcp3002-support) to preempt testing

ladyada, 2019-10-09:
awesome, thank you!

2bndy5, 2019-10-18:
@brentru I've had to alter the `AnalogIn.__init__()` a little to allow for the new MCP3002 class... I have a few questions:

1. Does the `negative_pin` parameter actually need to be saved to the object instance (`self._negative_pin`)? It seems like it is only used during instantiation to modify `self._pin_setting` accordingly, but then it is never used again.
2. What's up with the `__getitem__()`? I understand the concept of the function, but in `AnalogIn` it returns an element of `self._channels`. I see that `self._channels` is declared as a blank list when discerning the differential pin mapping, but then nothing is added to the that list. So wouldn't `__getitem__()` throw an `IndexError` exception? The way it currently written, `__getitem__` only applies for differential reads, and I see no `chan[<pin>]` in any of the examples. Is the `__getitem__()` meant to help mimic another type of analog input object like `analogio.AnalogIn`?

I don't mean to pick apart the code, I'm just trying to understand some design decisions.

brentru, 2019-10-21:
@2bndy5 Sorry for my delay in responding, it's exam-season for me.

> I don't mean to pick apart the code, I'm just trying to understand some design decisions.

You're not picking apart the code, you're improving it while adding a new chip! 

1. `_negative_pin` can be removed here (https://github.com/adafruit/Adafruit_CircuitPython_MCP3xxx/blob/master/adafruit_mcp3xxx/analog_in.py#L42). Instead, use `negative_pin` here: https://github.com/adafruit/Adafruit_CircuitPython_MCP3xxx/blob/master/adafruit_mcp3xxx/analog_in.py#L51

2. You can remove the `__getitem__` method there since AnalogIn's API doesn't mention it (see: https://circuitpython.readthedocs.io/en/4.x/shared-bindings/analogio/AnalogIn.html).

2bndy5, 2019-10-21:
I was hoping you'd say something like that. I've actually made quite a few "improvements" to all the MCP classes. Some with reduced computation on returned results, and others just avoid excessive Python garbage collection. [My branch has it all the commits](https://github.com/2bndy5/Adafruit_CircuitPython_MCP3xxx/tree/mcp3002-support). It is working for single ended reads on both MCP3008 & MCP3002, but I want to double check the "psuedo differential" reads work properly for both before creating a PR. Please continue focus on your exams 🙂

brentru, 2019-10-29:
Fixed by #20. Merged into `master` and released as `1.2.0`.

Closing out...

closed 2019-10-29 by brentru