adafruit/Adafruit_CircuitPython_MCP3xxx #11: Reading differential value for Pin 7 throws an error

yoomy3, 2019-03-08:
According to the MCP3008 datasheet, a proper way to calculate the differential value of Pin 7 should be Pin 7 value - Pin 6 value. (Table 5-2, page 19 in https://cdn-shop.adafruit.com/datasheets/MCP3008.pdf ).

Running this code snippet below throws an exception: 
~~~~
import time
import busio
import digitalio
import board
import adafruit_mcp3xxx.mcp3008 as MCP
from adafruit_mcp3xxx.analog_in import AnalogIn

spi = busio.SPI(clock=board.SCK, MISO=board.MISO, MOSI=board.MOSI)
cs = digitalio.DigitalInOut(board.D5)
mcp = MCP.MCP3008(spi, cs)

diff7 = AnalogIn(mcp, MCP.P7, MCP.P6)
print('Diff7: {}'.format(diff7.value)) 
~~~~

~~~~
Traceback (most recent call last):
  File "mcp3xxx-test2.py", line 13, in <module>
      print('Diff7: {}'.format(diff7.value))
  File "/usr/local/lib/python3.5/dist-packages/adafruit_mcp3xxx/analog_in.py", line 61, in value
    return self._mcp.read(self._pin_setting, is_differential=self.is_differential) << 6
  File "/usr/local/lib/python3.5/dist-packages/adafruit_mcp3xxx/mcp3xxx.py", line 86, in read
    command |= pin << 3
TypeError: unsupported operand type(s) for <<: 'str' and 'int'
~~~~
 
Looks differential channel mapping in MCP3008 class has a typo, (https://github.com/adafruit/Adafruit_CircuitPython_MCP3xxx/blob/master/adafruit_mcp3xxx/mcp3xxx.py)

MCP3008_DIFF_PINS = {
        (0, 1) : P0,
        (1, 0) : P1,
        (2, 3) : P2,
        (3, 2) : P3,
        (4, 5) : P4,
        (5, 4) : P5,
        (6, 7) : P6,
        **(6, 6) : P7**
    }

but somehow I was able to read the data on **Pin 7** using `AnalogIn(MCP.P6, MCP.P6).value` .

ladyada, 2019-03-09:
looks like a typo indeed! want to try fixing, testing if the typo fixes it, and submitting a PR? :)

yoomy3, 2019-03-12:
Hi Ladyada, we just submitted a pull request!

yoomy3, 2019-03-12:
Submitted a PR and merged, closing the issue.

closed 2019-03-12 by yoomy3

yoomy3, 2019-03-15:
Seems like `pip install adafruit-circuitpython-mcp3xxx` is still pulling v1.0.2. https://pypi.org/project/adafruit-circuitpython-mcp3xxx/ 
Any idea when will 1.0.3 be updated on Pypi?

ladyada, 2019-03-15:
@kattni @brentru ? can you check if something went wrong with the pypi deploy?

brentru, 2019-03-15:
@ladyada  @yoomy3  I just re-deployed the project. Now showing up on PyPi as v1.0.3

https://pypi.org/project/adafruit-circuitpython-mcp3xxx/1.0.3/

yoomy3, 2019-03-15:
@ladyada @brentru thank you for the quick response!

katurov, 2019-04-12:
Just checked all twice, but the question still here:

from adafruit_mcp3xxx.analog_in import AnalogIn
chan = AnalogIn(mcp, MCP.P0, MCP.P3)
i = chan.value

causes the same error:

Traceback (most recent call last):
  File "double-encoders.py", line 27, in <module>
    i = chan.value
  File "/home/pi/.local/lib/python3.5/site-packages/adafruit_mcp3xxx/analog_in.py", line 61, in value
    return self._mcp.read(self._pin_setting, is_differential=self.is_differential) << 6
  File "/home/pi/.local/lib/python3.5/site-packages/adafruit_mcp3xxx/mcp3xxx.py", line 86, in read
    command |= pin << 3
TypeError: unsupported operand type(s) for <<: 'str' and 'int'

So I've checked versions:
pip3 freeze | grep adafruit
adafruit-circuitpython-busdevice==2.2.9
adafruit-circuitpython-mcp3xxx==1.0.3

So I've tried to update:
pip3 install -U adafruit-circuitpython-mcp3xxx
[...]
Successfully installed Adafruit-Blinka-1.2.8 Adafruit-PlatformDetect-1.0.0 Adafruit-PureIO-0.2.3 RPi.GPIO-0.6.5 adafruit-circuitpython-busdevice-2.2.9 adafruit-circuitpython-mcp3xxx-1.0.3 rpi-ws281x-4.1.0 spidev-3.4 sysv-ipc-1.0.0

BUT the error is still here.

caternuson, 2019-04-15:
Your error looks like it's coming from a larger program called `double-encoders.py`, so it may be something subtle there that's causing `pin` to become a `string` instead of an `int`. Can you recreate the error using one of the examples?
https://github.com/adafruit/Adafruit_CircuitPython_MCP3xxx/tree/master/examples

katurov, 2019-04-19:
Thanks for help, today I cannot reproduce the error even in my program (but example works just fine). Sorry that borrow you.