adafruit/Adafruit_CircuitPython_MCP3xxx #12: Values from pins return 16 bits, instead of 10 bits

yoomy3, 2019-03-09:
I am expecting the digital values read from MCP3008 to be between 0 and 1024.

> The MCP3008 is a low cost 8-channel 10-bit analog to digital converter. 

Digital Output Code  = (1024 * V_in) / V_ref
(Equation 4-2, page 17, https://cdn-shop.adafruit.com/datasheets/MCP3008.pdf)

I realized even in the example in the official document, MCP3008 throws values between 0 and 65535, which is 16 bits, not 10 bits.

[image]
(https://learn.adafruit.com/mcp3008-spi-adc/python-circuitpython)

In order to retrieve correct values, I manually had to shift 6 bits down `channel.value >> 6` but shouldn't it be handled in the library, returning values in 10 bits?

ladyada, 2019-03-09:
we normalize to 0-65535 to match other ADCs (either on-chip or extenral) that way no matter which one you use, you'll get 16 bits of data. the ADC voltage is what most people want anyhow :)

@kattni perhaps document in the learn guide? this isn't a bug tho, its a FEATURE :D

closed 2019-03-09 by ladyada

yoomy3, 2019-03-12:
Thanks for the reply!
Just a thought, if it's a feature then shouldn't there be a parameter to choose whether the returned value is in 10 bits or 16 bits? I was quite confused to get large values, and manually shifting 6 bits down does not look like an ideal workflow for 10-bit ADC.

ladyada, 2019-03-12:
we prefer to keep it this way and just make sure its documented!

yoomy3, 2019-03-12:
Got it. Thanks :)

caternuson, 2019-03-15:
We could add another parameter that would be actual ADC bit level. The most difficult thing being deciding what it should be called.

ladyada, 2019-03-15:
cater, i don't want to do that - complicates the API too much :/
i'd rather people have helper functions if they need

cbernet, 2019-06-10:
Sorry for commenting on this closed PR, but I just got hit by this one as well. 

This feature still does not appear to be documented here https://learn.adafruit.com/mcp3008-spi-adc/python-circuitpython#. 

What about simply adding in the example code: 

```python
# the raw ADC value is encoded on 16 bits to match other ADCs 
# converting to 10 bits as the MCP3008 only has 10 bits
print('Raw ADC Value: ', chan.value >> 6)
print('ADC Voltage: ' + str(chan.voltage) + 'V')
```

This would have saved me quite a bit of time :-) 

Thanks a lot for all the material you're offering (hardware and documentation)

ladyada, 2019-06-10:
@kattni please update the guide!

tukno, 2019-07-12:
Sorry for commenting on this closed issue, but the same thing happened to me until I figured it out.

Please update the guide :)

siddacious, 2019-07-12:
Guide updated!