adafruit/Adafruit_CircuitPython_MCP3xxx #18: ReadTheDocs hasn't been building releases since v1.0.1

2bndy5, 2019-10-14:
I hate to be that guy, but RTD doesn't seem to be automatically triggering docs builds. I can't help fix this through a PR. This is on whoever is in charge of [the RTD project for this library](https://readthedocs.org/projects/adafruit-circuitpython-mcp3xxx/builds/)

2bndy5, 2019-10-14:
Never mind. Never saw RTD take hours to rebuild. Docs are live again

brentru, 2019-10-14:
@2bndy5 No, you're correct. I manually started a  RTD build and re-sync'd the webhook for this repo's RTD.

brentru, 2019-10-14:
@2bndy5  Looks like it updated OK, pls re-open if you find it going out of sync again.

closed 2019-10-14 by brentru

2bndy5, 2019-10-14:
I have also had a webhook problem reported on RTD, but functioning correctly (as reported via GitHub). I think this may be more of a RTD issue... Only happened on a repo that I don't "own" (repo is from a GitHub organization account).