adafruit/Adafruit_CircuitPython_MCP3xxx #21: Incorrect voltage calculation

VyReN, 2019-10-28:
I think the final line of analog_in.py is wrong.  Even when I drive 3.3 volts to an input, the value returned is 3.29682...

`return (self.value * self._mcp.reference_voltage) / 65535`

When it should be:

`return (self.value * self._mcp.reference_voltage) / 65472`

The highest level I'm getting from the MCP3008 is 65,472, not 65,535.  This makes sense, as the MCP3008 outputs 0 to 1,023 (for a total of 1,024 values).  1,023 * 64 = 65,472

Am I looking at this right?

ladyada, 2019-10-28:
ADCs are not 100% rail to rail - there is always some error!

closed 2019-10-28 by ladyada

kevinjwalters, 2024-09-02:
@VyReN The library used to return `0b1111111111000000` (65472) for the upper value. This changed in #44 to give all 1's, i.e. 65535. If you have code using 65472 and have picked up a recent library you'll need to accomodate that.