adafruit/Adafruit_CircuitPython_MCP3xxx #27: AttributeError: module 'board' has no attribute 'SCK'

grektor, 2020-09-27:
As follow-up to the issue #26:

Side note: The upgraded code has started to work for me with one FSR sensor, but the the printed voltage (result) was less sophisticated compared to one generated by the old code.
The real issue is that after the installation of few additional pyhton packages, the new code has stopped working:

spi = busio.SPI(clock=board.SCK, MISO=board.MISO, MOSI=board.MOSI)
AttributeError: module 'board' has no attribute 'SCK'

I have not find any similar like that recorded, but I hope you can advise! Thank you in advance!

caternuson, 2020-09-27:
Did you pass the Blinka post-install check?
https://learn.adafruit.com/circuitpython-on-raspberrypi-linux/installing-circuitpython-on-raspberry-pi#blinka-test-3030038-17

You may have installed the wrong board package:
https://learn.adafruit.com/circuitpython-on-raspberrypi-linux/faq-troubleshooting#faq-3055911

closed 2020-09-28 by caternuson

grektor, 2020-10-03:
Thank you, it worked! Actually, "board" should not be installed!