adafruit/Adafruit_CircuitPython_MCP3xxx #28: conceptual questions

grektor, 2020-10-04:
**Question #1:** By printing out digital (FSR) sensor data (using MCP3008) together with statement "datetime.datetime.now()", what do I really see? Does the signal reflect the status at the moment of SENSING or that at the moment of RECORDING (moment of sensing is earlier than moment of recording)?

**Question #2**: In connection to the question above and additionally using the statement "time.sleep(0.1)", the sampling rate is ~10/second and the results are displayed very quickly on the display. However with statement "time.sleep(0.0)", the sampling rate is ~230/second and the results are displayed ~1 second later (once I pressed the sensor pad) on the monitor. If I would like to precisely map the FSR sensor data to other time-series using "datetime" unique key, should I shift the result by ~1 second earlier to reflect the reality, or actually the sensor data and associated "datetime" are linked correctly (irrespective of the sampling rate) but the 1 second delay is only visible in real-time display mode?

Thank you for your help!

ladyada, 2020-10-04:
hi these are better posted to the tech support forum, it isnt a bug with the library
thanks!

closed 2020-10-04 by ladyada

tannewt, 2020-10-05:
The tech support forum is here: https://forums.adafruit.com/