adafruit/Adafruit_CircuitPython_MCP3xxx #29: gpio 8 being selected as CS (chip select) alongside programmed pin

pedro-dreyer, 2020-10-24:
I noticed that independently of which GPIO pin I select to be the CS(chip select) it will also activate GPIO pin 8. Is this the intended behaviour?

tannewt, 2020-10-26:
Are you on Raspberry Pi? This could be a bug in the Blinka library, not this one.

eleimt, 2020-11-01:
Hello.
You find solutions?
How can I use CE1 (GPIO7) instead of CE0 (GPIO8)?

pedro-dreyer, 2020-11-01:
@tannewt Yes, I am on raspberry pi. Well my debugging skill are very limited so I don't really know if I can track the source of the bug. Thanks for the input anyways 

@eleimt I think you just need to select the cs pin as shown in the example, but changing the 5 for a 7: 

`cs = digitalio.DigitalInOut(board.D7)`

Note that my problem was not selecting a particular GPIO pin as CS, but that regardless of which i choose GPIO8 would act as CS as well.

makermelissa, 2021-01-25:
Hi, I've looked into this before and found this is a behavior of the linux kernel that likes to automatically twiddle with the CE0 port, which is why we like to use a different GPIO when we have issues on SPI. In practice, most of the time the twiddling works fine, but I've noticed this issue can affect certain boards and in particular on the RA8875. I ended up recommending using a different GPIO (see wiring in https://learn.adafruit.com/ra8875-touch-display-driver-board/python-wiring-and-setup). Unfortunately, it's a quirk of linux and there isn't much we can do.

closed 2021-01-25 by makermelissa