adafruit/Adafruit_CircuitPython_MCP3xxx #34: spi1 instead of spi0

Schmediii, 2021-04-01:
Sorry that i write this in the Issues but i have a question.
I want to use your code to read analog inputs with a raspi with the MCP3008. But not with the spi0 but with the spi1, is this possible and how can i change this? For example in the code of mcp3xxx_mcp3008_single_ended_simpletest.py.
Thank you for your help.

jposada202020, 2021-04-20:
@Schmediii Please use the Adafruit Discord Channel for these kind of questions, you will get more visibility and probably more answers. or the Adafruit Forums if you like. Thanks

benocd, 2022-01-07:
Hey @Schmediii did you find the solution for this?

tekktrik, 2022-10-07:
Yes, it is possible using Blinka on a Raspberry Pi.  For more help or information, please use the previously mentioned resources.  Thanks!

closed 2022-10-07 by tekktrik