adafruit/Adafruit_CircuitPython_MCP3xxx #41: Please add MCP3208 support.

jpnielsen, 2023-01-20:
"may later support MCP320x as well"

So, Eben Upton and co has made 10.000.000 RP2040, that needs external ADC's to be useful beyond 8 bit.
https://datasheets.raspberrypi.com/rp2040/rp2040-datasheet.pdf#errata-e11

I found this library, but ideally want to use MCP3208 to get the resolution I think I need.
The application is a simple 2-joystick(4 pots) + 4 pots MIDI device for sending analog values as 14-bit MIDI CC MSB/LSB messages.

So please, If you have the time, update this to include MCP3208 support.

Kind regards
Jens Peter Nielsen

kevinjwalters, 2024-09-01:
I need this too, I'm just having a look...

kevinjwalters, 2024-09-27:
@jpnielsen The RP2040 ADC isn't too bad. Have a look at the tests I did some casual testing a while back [Adafruit Forums: Feather ADC comparison including 2.6V limited ESP32-S2](https://forums.adafruit.com/viewtopic.php?f=60&t=174020) and if you look really closely you can see the fault.

I've just been doing a bit more testing and the noise is mostly within +/- 4LSB which is better than the nRF52's +/-8LSB which you can see in [Exploring and Reducing ADC Noise on Adafruit CLUE (Nordic NRF52840)](https://www.instructables.com/Exploring-and-Reducing-ADC-Noise-on-Adafruit-CLUE-/).

I've also checked and you can get pretty good values out of the RP2040 by removing the offset and taking average of ten samples. I'd say the main advantage of external ADCs like MCP3208 is you can take a _single_ sample and it will be low noise plus you get a decent number of channels. The RP2040 can be frustrating with its 4, or 3 if using Pi Pico (W).