adafruit/Adafruit_CircuitPython_MCP3xxx #42: OSError running sample script for MCP3008 on Libre Computer aml-s905x-cc "Le Potato"

anguselheimStudios, 2023-04-18:
I am running into an OSError trying to run sample code for MCP3008 on the Libre Computer aml-s905x-cc "Le Potato". I have made slight changes to sample code, changing the pin numbers, putting the print statements in a loop, and I've tried busio and extendid bus with the same results . 

**Here is the script I am trying to run:**

```
import time
import busio
import digitalio
import board
import adafruit_mcp3xxx.mcp3008 as MCP
from adafruit_mcp3xxx.analog_in import AnalogIn
from adafruit_extended_bus import ExtendedSPI as SPI

# create the spi bus
#spi = busio.SPI(clock=board.P23, MISO=board.P21, MOSI=board.P19)
spi = SPI(0, 0)

# create the cs (chip select)
cs = digitalio.DigitalInOut(board.P22)

# create the mcp object
mcp = MCP.MCP3008(spi, cs)

# create an analog input channel on pin 0
chan = AnalogIn(mcp, MCP.P0)

while True:
	print('Raw ADC Value: ', chan.value)
	print('ADC Voltage: ' + str(chan.voltage) + 'V')
	time.sleep(5)

```
**Here is the error:**
```
angus@aml-s905x-cc:~/scripts$ python3 mpc3008_test.py 
Traceback (most recent call last):
  File "/home/angus/scripts/mpc3008_test.py", line 23, in <module>
    print('Raw ADC Value: ', chan0.value)
  File "/usr/local/lib/python3.10/dist-packages/adafruit_mcp3xxx/analog_in.py", line 60, in value
    self._mcp.read(self._pin_setting, is_differential=self.is_differential) << 6
  File "/usr/local/lib/python3.10/dist-packages/adafruit_mcp3xxx/mcp3xxx.py", line 91, in read
    spi.write_readinto(self._out_buf, self._in_buf)
  File "/home/angus/.local/lib/python3.10/site-packages/busio.py", line 396, in write_readinto
    return self._spi.write_readinto(
  File "/home/angus/.local/lib/python3.10/site-packages/adafruit_blinka/microcontroller/generic_linux/spi.py", line 133, in write_readinto
    data = self._spi.transfer(list(buffer_out[out_start : out_end + 1]))
  File "/home/angus/.local/lib/python3.10/site-packages/Adafruit_PureIO/spi.py", line 420, in transfer
    ioctl(self.handle, SPI._IOC_MESSAGE, spi_ioc_transfer)
OSError: [Errno 22] Invalid argument

```

**System Info:**
```
angus@aml-s905x-cc:~/scripts$ neofetch
           `:+shmNNMMNNmhs+:`              angus@aml-s905x-cc 
        .odMMMMMMMMMMMMMMMMMMdo.           ------------------ 
      /dMMMMMMMMMMMMMMMmMMMMMMMMd/         OS: Ubuntu MATE 22.04.2 LTS aarch64 
    :mMMMMMMMMMMMMNNNNM/`/yNMMMMMMm:       Host: aml-s905x-cc 
  `yMMMMMMMMMms:..-::oM:    -omMMMMMy`     Kernel: 6.1.21-04467-gc1663d8fbf25 
 `dMMMMMMMMy-.odNMMMMMM:    -odMMMMMMd`    Uptime: 9 hours, 14 mins 
 hMMMMMMMm-.hMMy/....+M:`/yNm+mMMMMMMMh    Packages: 2404 (dpkg), 9 (snap) 
/MMMMNmMN-:NMy`-yNMMMMMmNyyMN:`dMMMMMMM/   Shell: bash 5.1.16 
hMMMMm -odMMh`sMMMMMMMMMMs sMN..MMMMMMMh   Resolution: 1680x1050 
NMMMMm    `/yNMMMMMMMMMMMM: MM+ mMMMMMMN   DE: MATE 1.26.0 
NMMMMm    `/yNMMMMMMMMMMMM: MM+ mMMMMMMN   WM: Metacity (Marco) 
hMMMMm -odMMh sMMMMMMMMMMs oMN..MMMMMMMh   Theme: Yaru-MATE-dark [GTK2/3] 
/MMMMNNMN-:NMy`-yNMMMMMNNsyMN:`dMMMMMMM/   Icons: Yaru-MATE-dark [GTK2/3] 
 hMMMMMMMm-.hMMy/....+M:.+hNd+mMMMMMMMh    Terminal: tilix 
 `dMMMMMMMMy-.odNMMMMMM:    :smMMMMMMd`    CPU: (4) @ 1.512GHz 
   yMMMMMMMMMms/..-::oM:    .+dMMMMMy      Memory: 1573MiB / 1924MiB 
    :mMMMMMMMMMMMMNNNNM: :smMMMMMMm:
      /dMMMMMMMMMMMMMMMdNMMMMMMMd/                                 
        .odMMMMMMMMMMMMMMMMMMdo.                                   
           `:+shmNNMMNNmhs+:`

angus@aml-s905x-cc:~/scripts$ pip3 list | grep adafruit
adafruit-circuitpython-busdevice 5.2.4
adafruit-circuitpython-mcp3xxx   1.4.14
adafruit-circuitpython-pixelbuf  2.0.0
adafruit-circuitpython-requests  1.13.0
adafruit-circuitpython-seesaw    1.11.7
adafruit-circuitpython-typing    1.9.0
adafruit-extended-bus            1.0.2
angus@aml-s905x-cc:~/scripts$ pip3 list | grep Adafruit
Adafruit-bitfield                1.5.1
Adafruit-Blinka                  8.16.1
Adafruit-GPIO                    1.0.3
Adafruit-PlatformDetect          3.42.0
Adafruit-PureIO                  1.1.10

```

**I was made aware of this issue by this forum post:**

https://hub.libre.computer/t/problems-accessing-mcp3008-via-spi-invalid-argument/1844

anguselheimStudios, 2023-04-22:
At Libre Computer's suggestion I have written a Device Tree Overlay for the MCP3008 which now shows up as an iio device, and can be read using the open() function. It seems to work, however think I cooked my MCP wiring it improperly. I'm getting some strange results now (see the thread linked in my initial post). 

I'm going to order a replacement and test it. If it works properly, and I can manage not to melt this one, I can try to add support for iio devices to this module.

Is that something anybody is interested in or should I not bother?

anguselheimStudios, 2023-04-28:
I have the MCP3008 working as an iio device. My test code reads the device with the open() built-in.

```

def read_mcp3008(channel=0, device=1):
	'''Reads MCP3008 connected as an iio device.Takes a channel number (0-7)
	and iio device number as arguments. Returns an integer from 0-1023'''
	channel_path = f"/sys/bus/iio/devices/iio:device{device}/in_voltage{channel}_raw"
	with open(channel_path, 'r') as f:
		return(f.read())

 ```

Since I haven't heard anything I'm just going to go for it.

I plan on creating an alternate "mcp3xxx-iio" class and adding an "iio_device=-1" argument to the "mcp3008" class. This argument will default to -1 for the regular "mcp3xxx" class. If it is given a non-negative value it will specify the iio device number and inherit from the new "mcp3008-iio" impostor class. 

I do have a few questions, as I'm new to github:
1. How should I handle the Author(s) line?
2. Is there a style guide?
3. I don't have the other 2 devices covered by this module. Is there someone else available to test them?
4. Would you rather I just keep this to myself and my potato people?

I would appreciate any input or criticism.

Thanks,
Angus

tekktrik, 2023-04-28:
It looks like Blinka is using the generic Linux code, is that normal for the Libre Computer aml-s905x-cc "Le Potato", or should it be using another implementation / pin definition?

anguselheimStudios, 2023-04-28:
I was told when a Linux driver exists you should use it. Either way, it's not working on the Le Potato in it's present form, so somethings got to give. 

The way it's been looking, I'm just going to make a separate set of classes for use with these chips iio devices, since I can't really implement my plan and preserve backwards compatibility. 

At the end of the day I want it to work, and to be able to keep my code portable for use with my microcontrollers with as few changes as possible.